TampereHacklab/Inventaario #2: NFC-ovenavaus

vranki, 2020-12-27:
Asenna NFC-ovenavaus ovelle

vranki, 2021-02-17:
Vaatii ilmeisesti koteloinnin ja paikalleen asennuksen.

pinq-, 2021-12-14:
- [x] Asennettu ja koteloitu
- [ ] Koodin päivittäminen niin, että sisä ovi pysyy kauuempaa auki.
- [ ] Näytön tarpeen pohtiminen? Jos tykkää glitch taiteessa, nii se näyttää hienolta kosteella säällä, mutta pidemmän päälle saattaa hajoittaa laitteen.
- [ ] Enemmän kuin yhden avaimen rekistörinti?

vranki, 2023-02-05:
NFC-avaus ei toimi tällä hetkellä, lukija ei reagoi mitenkään. Ei tietoa tarkemmasta syystä.

vranki, 2023-02-17:
Toimi taas eilen. Ei tietoa onko joku korjannut vai mistä johtuu.

vranki, 2023-02-17:
@pinq- teetkö omat issuet noista rukseista. Avaus toimii, nuo ovat jatkokehitysideoita. Suljen tämän tiketin.

Ps. Enemmän kuin yhden avaimen rekisteröinti tekisi helpoksi avainten tekemisen kavereille ja sitä ei mun mielestä haluta.

closed 2023-02-17 by vranki