TampereHacklab/Inventaario #6: Wood Big CNC

vranki, 2021-02-04:
Rikki on

vranki, 2021-02-04:
Tämä on testi. Onkohan se oikeasti rikki?

sbeach92, 2021-02-04:
Isoon cnc:hen vaihdettu laakerit. Ensimäisillä ajoilla tuntu lämpenevän.
-Muutaman työtunnin jälkeen lämpenieminen vähäisempää.
-Vältetään täysiä kierroksia mahdollisuuksien mukaan
-Seurataan väheneekö lämpeneminen

pinq-, 2021-12-14:
Välihuomio: Kone lämpenee edelleen vähän. Ei kai olisi hyvä ajaa liian pitkää ohjelmaa. Syytä ei ole tiedossa :/

tswfi, 2021-12-14:
Sanoisin että suljetaan tämä pois ja lisätään wiki sivulle note että seurailee lämpöjä varsinkin jos ajaa pitkiä rupeamia

pinq-, 2021-12-16:
En välttämättä sulkisi tätä, koska ongelma ei ole poistunut ja se on edelleen ongelma, eikä ominaisuus. Et jos joku vaikka ottaisi tästä kopin, jos se on täällä

sbeach92, 2021-12-16:
Mun on ainakin vaikea sanoa onko ongelmaa vai ei.

Standardi oikosulkumoottori rakennetaan niin että ambient 40°C asteessa se lämpiää nimelliskuormallaan 80°C eli käämit kestää 120°C. Kiinalaisesta moottorista en osaa sanoa. 

Tyhjäkäynnillä moottori lämpesi ulkokuoresta noin 70°C ja nimenomaan käämien kohdalta. Vesijäähdytys on vain alimmaisten laakerien jäähdyttämiseksi.

Taajuusmuuttajan parametritkin voivat olla pielessä. Nimellisjännite, nimellistaajuus ja nimellisvirta vaikuttavat kaikki moottorin lämpenemiseen. Ohjataanko moottoria V/f vai vuovektori tyyppisesti? Parametrejä vaan on vaikea asettaa oikein kun moottorista ei ole dataa.

tswfi, 2022-06-07:
Suljen tämän pois, ei ole tullut kommentteja vähään aikaan ja kyllähän moottorit aina vähän lämpiää.