TampereHacklab/Inventaario #9: Metallijyrsimen käyttöönotto

vranki, 2021-03-03:
* Laitetaan uudet tassut alle ja se säädetään vaateriin
* Putsaus vuosikymmenten likakerroksesta 
* Perushuollot 
* Kuularuuvin vaihto

Do the needful.

akusav, 2021-03-19:
- sähköt (suurempi sulake)
- paineilman vetäminen
- karahihnan vaihto
- suodattimen vaihto
- näytön huolto
- x-ruuvin huolto/vaihto

akusav, 2021-04-21:
Tehty:
- sähköt
- paineilma
- näyttö
Kesken:
- karahihnan vaihto ja kiristys
- suodattimien vaihto?
- x-ruuvin huolto/vaihto
- kotelon puhdistus/maalaus

akusav, 2022-03-06:
- Työstöpöydän liukupintojen liimaaminen

Cromfel, 2022-05-22:
Muutin taskit täppälaatikoiksi

tswfi, 2022-06-09:
Sähköjä pitää laittaa uudestaan

pinq-, 2022-11-25:
Seuraava vaihe on koteloseinien maalminen. Pöytä on Hiekkaranta_92 työn alla.

sbeach92, 2024-02-06:
- [ ] Asennus tassut sorvattu
- [ ] Koneen asennus tassuille
- [ ] Tassujen liimaus lattiaan
- [ ] pöydän liukupintojen kaavaus

Johde voitelu
- [ ] Vaihda annostelija X/Y
- [ ] Puhalla linjat X/Y
- [ ] Tarkasta että putket on kuvan mukaan
- [ ]  Kiristä letkuliittimet
- [ ] Vaihda annostelijat Z]