TampereHacklab/Inventaario #11: Peräkärryn valot täytyy uusia

vranki, 2021-03-28:
Peräkärryn valot toimii "miten sattuu" (tilanne pollattu Q4/2020 viimeksi).

Valot ovat niin huonossa kunnossa, että ne pitäisi kokonaan rakentaa uusiksi.
Esim biltemasta saa valmiita LED-moduleja tähän käyttöön.

Kaapelin kunto kannattaa myös tarkastaa, voisi olla hyvä samalla vaivalla vaihtaa kaikki 
pistokkeesta taaksepäin. Osien hankinta luonnollisesti labin piikkiin.

vranki, 2021-04-06:
Päivitys pääsiäisenä: Sain valot elvytettyä niin että toimii:

 - 2 etuvaloa koko ajan
 - 2 vilkkua (vilkkuvat kun autossakin ja samalla puolen)
 - 1 punainen takavalo koko ajan
 - 1 punainen jarruvalo (syttyy kun painaa jarrua)

Kärry on siis käyttökunnossa, mutta suosittelen testaamaan valot ennen ajoa. Jätetään tiketti
auki koska kyllä nuo olisi syytä vaihtaa LEDeiksi.

vranki, 2021-05-04:
Budjetoitu 100€ 4.5 yhteisömiitissä. Jonkun pitäisi hakea tarvittavat osat seuraavaksi.

Miggov, 2021-11-07:
Laitetaas tänne ylös että ledivalot ei pelaa yhteen kaikkien autojen kanssa millään adapterilla vaan alkavat vilkkua kun auto testaa "polttimoiden" toimivuutta

Miggov, 2021-11-07:
Sit myös adapteri, joka muuntaa kärryn 13pin liittimen 7 pinniseksi ei ole hyvä, töpselistä roikkuva massa on tässä kohtaa niin iso että valot alkavat hetken ajon jälkeen vilkkumaan. Toisinpäin oleva adapteri lukittuu autoon paremmin ja on hyvä. Kärryyn voinee toteuttaa myös 2 piuhaa jos sen 13pin liittimen sinne haluaa (tarvitaanko?)

Miggov, 2021-11-07:
> 
> 
> Laitetaas tänne ylös että ledivalot ei pelaa yhteen kaikkien autojen kanssa millään adapterilla vaan alkavat vilkkua kun auto testaa "polttimoiden" toimivuutta

Eli tldr ei ledejä jos halutaan että kaikki joilla on koukullinen auto voi tota vetää

akusav, 2022-03-20:
- palataan asiaan kevään tullen, vaatii selvittämistä
- rekisterivalon kaluste pitää uusia
- viimeisin testi takavalot eivät toimi, jarruvalot ja vilkut toimivat sekä äärivalot edessä

Rakkaustermari, 2022-10-11:
Kaikki valot toimivat. Kärryssä on maadoitus ongelmia johtuen ruosteesta polttimoiden kontaktipinnoilla. Electrospray ja hiomapaperi yleensä tehoaa...

pinq-, 2022-11-23:
Tämä on kai korjattu ja laite lienee toimivan. Avataan uusi issue, jos tilanne muuttuu.

closed 2022-11-23 by pinq-