TampereHacklab/Inventaario #13: Kemiahuoneen viimeistely ja käyttöönotto

akusav, 2021-04-21:
- seinien pintakäsittely
- rakojen tiivistys
- kärypoistot

pinq-, 2021-12-14:
- [ ] Pölyn/käryn poiston tekeminen. 
   - [ ] Suunnitelman tekeminen ilmanpoistolle
   - [ ] Vaatii jonkun "uskalluksen" aloittaa tekemään. 
   - [ ] Vetokaapit vaativat oman ilmanvaihdoin myös. (Käytännössä imu ilma imettäisiin niiden kautta huoneesta)
   - [ ] Materiaalin hankkiminen
- [ ] Uudelleen järjestely
- [x] Raot tiivistetty
- [x] Ulkopinnat käsitelty

akusav, 2022-03-06:
Maalaamo tarvii minimissään:
- keittiölle menevien vesiputkien kiinnityksen seinään (roikkuvat nyt hasardisti miten sattuu)
- suodattimen poistokanavaan, alkuun varmaan riittää puupuolellakin käytössä oleva suodatinkangas (ettei maalisumu/pöly mene masiinaan)
- suojamaadoitettu jatkojohto poistopuhaltimelle
- tilan raivaus

akusav, 2022-03-06:
- Poistopuhaltimen käyttökytkimen asentaminen

pinq-, 2022-11-23:
Tämä lienee "viimestelty". Seuraavat askeleet voidaan lisätä seuraaviksi issukoiksi

closed 2022-11-23 by pinq-