TampereHacklab/Inventaario #14: 3D-tulostin Prusa

akusav, 2021-04-21:
"3d-printterissä taitaa olla z-akselissa ja/tai kalibroinnissa häikkää. mikään ei tarttunut ja 'preamblestakin' tuli morsekoodia. tulostuslevy on jo osittain 'kaiverrettu'."

Miggov, 2021-08-16:
Kesällä toimi, mikähän tilanne mahtaa olla nyt? Printti ei vaikuttanut tarttuvan taas hiljattain testaillessa ja lämmitinelementin silikonisuojus näyttäisi roikkuvan, voisiko tuo repiä printit irti?

Miggov, 2021-08-16:
En saa otettua taskia itelleni mutta otan tän nyt hoitoon ja tilaan uusia silikonisukkia. Nuo voisi yrittää köyttää jollain mielellään ruostumattomalla jekkulangalla kiinni etteivät pääse putoamaan ja tuhoamaan printtiä tai pahimmillaan blokkaamaan suutinta

pinq-, 2021-12-14:
Tämä on korjattu ja pävitetty. Tämän voi sulkea

closed 2021-12-14 by tswfi