TampereHacklab/Inventaario #17: Loungen uudelleen rakennus järkeväksi

pinq-, 2021-12-14:
Oleskelutilan tekeminen järkevämmäksi.

- Vaatisi selkeämmän rakenteen/järjestyksen sohvien ja pöytien kanssa.
- Ehdotettu vanan loungen tapaista baaripöytää, jossa olisi integroidut pistorasiat. Voisi olla myös S:n muotoinen, jollin olisi enemmän työtilaa.

vranki, 2021-12-15:
Tämä issue pitäisi määritellä tarkemmin tai pilkkoa osiin. Pitäisi olla selkeä yksittäinen tehtävä josta voidaan selvästi tietää koska se on valmis. Ehdottaisin että ensimmäinen tehtävä on palaveri labilla jossa keksitään konkreettisia tehtäviä.

Cromfel, 2022-05-22:
Pilkottu alitehtäviin tiketin siällä, jos kokonaisuudet suurempia, voi eritellä ticketeihin

kengu2, 2022-05-22:
Loungea on taidettu pyöräyttää uuteen uskoon noin kahdesti viime joulukuun jälkeen. Tästä pitäisi varmaankin olla jotain tarkempaa suunnitelmaa, pohjapiirrosta tai jotain.

tswfi, 2022-06-09:
Tässä muodossa tämä taski ei etene.
Suljen pois.

closed 2022-06-09 by tswfi