TampereHacklab/Inventaario #23: Elektroniikkatyötilan järjestely

akusav, 2022-03-06:
Komponenttien:
  - [ ] inventointi
  - [ ] järjestely

Sandplum, 2023-09-03:
Equipment from the Electronics area was sorted and a pallet was filled, moving some of the equipment to the central pallet storage area.  The equipment has been documented on the wiki.  https://wiki.tampere.hacklab.fi/tyotilat/elektroniikkatila

[image]

314Sami, 2025-03-08:
This work has been restarted in 2025 and the current plan and status is listed below:
This guidance will evolve with the Process and remember to check it regularly. Ask @314Sami, @Sandplum and other active people for more details.

NEEDED:
- [x] Physically install PC to storage room
- [ ] Install component database program
- [ ] Build photo booth
- [x] Bring ring light for taking pictures
- [x] Bring colored sticky note paper, pens, tape 

STEP 1: (ongoing)
- [ ] Approximate each component type amount to allocate how many shelf meters / cubic meters are needed for each type
- [ ] Collect similar components together, but avoid mixing already sorted ones, there are ziplock bags and boxes to keep them separate.
STEP 2:
- [ ] Take one type of component (for example all potentiometers, or all LEDs, or all IC's, ...)
- [ ] Sort them by values
- [ ] Take a good picture of the amount of single value (are there 1, 5, 10, 50, 100+ of it)
- [ ] Take a detailed picture of a single component or arrangement of few so different angles are covered
- [ ] Mark the subsets with right colored note sticker + tape + initials + date + image serial number so others can know what has been done
- [ ] Upload the pictures in Nextcloud
- [ ] Return the marked items in storage but do not let them mix

STEP 3:
- [ ] Search component datasheets and/or most important values
- [ ] Input data and pic and datasheet to database

STEP ?:
- [ ] Find logical places to all components
- [ ] Mark each "place" with a alphanumeric tag
- [ ] Mark each component location to database with said tag
- [ ] Print sticky labels to boxes, drawers etc