TampereHacklab/Inventaario #35: Uuden mittalaitepöydän asennus

vranki, 2022-06-10:
Mittalaitepöytä on nyt labilla:

[image]

Tälle pitäisi päättä asennuspaikka ja suorittaa asennus. Pöytä painaa aika paljon joten tarvinee jonkinlaiset
custom jalat. Pöytään voi integroida oskilloskooppeja sun muita laitteita. Tyhjiä etupaneeleita tuli mukana N
kpl.

Ennenkuin työ on valmis, pöytään tulee tehdä kilpi (tarratulostetimella jos laiskottaa) jossa lukee: "Lahjoittanut Tekn. Tri Tiiti Kellomäki 8.6.2022".

Kuvia mukana tulleista varusteista:

[image]

[image]

akusav, 2023-02-28:
Johonkin tää katos mutta mihin? Voidaan varmaan sulkea tää issue?

pinq-, 2023-07-04:
Joo. Ei ole näkynyt.

closed 2023-07-04 by pinq-