TampereHacklab/Inventaario #36: Paloturvallisuuden parantaminen!

akusav, 2022-07-28:
22.7 alkaen hacklabilla keskitytään paloturvallisuuden parantamiseen palotarkastajan huomioiden pohjalta. Tästä syystä TULITÖITÄ EI SAA TEHDÄ ja PUUTYÖTILAT OVAT POISSA KÄYTÖSTÄ toistaiseksi. Yhteisöä informoitu sähköpostitse ja pikaviestinkanavilla asiasta.

- [ ] tulityötilan merkintä
- [ ] ketjutettujen jatkojohtojen poistaminen
- [ ] sammutuskaluston merkitseminen
- [ ] tulityötilaan hankitaan 2 kpl:tta, minimissään luokan 43A 183BC palosamuttimia
- [ ] nestekaasupullot lastauslaiturille häkkiin tms. sillon kun ei käytetä
- [ ] nestekaasupullojen merkitseminen
- [ ] ylimääräisten nestekaasupullojen poistaminen hacklabin tiloista
- [ ] puutyötilan pölyongelman korjaaminen, aloitetaan siivoamalla tila hyvin, tulevassa kuukausimiitissä pohditaan miten tilannetta voitaisiin parantaa pysyvämmin
- [ ] hätäpoistumisteiden merkitseminen
- [ ] kulkuväylät tyhjäksi; sinne kuulumattomat tavarat hävitetään
- [ ] tilojen yleisen siisteyden parantaminen
- [ ] kemia huoneen siivoaminen ja järjestely
- [ ] ongelmajätteet nekalan jäteasemalle
- [ ] sopivien palovaroittimien hankinta ja asennus
- [ ] jäsenen ohjeen päivittäminen paloturvallisuuden osalta https://wiki.tampere.hacklab.fi/jasenen_ohje#tulipalot
- [ ] tulevassa kuukausimiitissä mietitään turvallisuusvastaavan vakanssin perustamista
- [ ] yhdistyksen hallitus selvittää puupuolen sähköasennusten parantamiseen tarvittavia toimia
- mitä muuta? auta meitä tekemään hacklabistä turvallisempi paikka ja osallistu!

Mitä nopeammin epäkohdat saadaan korjattua, sitä nopeammin tilat ovat jälleen meidän kaikkien käytettävissä. Listausta ja tehtyjä toimenpiteitä tarkastellaan huoltomiehen kanssa 1.8 hacklabillä.

tswfi, 2022-07-28:
[image]

tswfi, 2022-07-28:
[image]

tswfi, 2022-07-28:
[image]

tswfi, 2022-07-28:
[image]

[image]

tswfi, 2022-07-28:
[image]

akusav, 2022-08-03:
[image]

akusav, 2022-08-03:
[image]

akusav, 2022-08-03:
[image]

akusav, 2022-08-04:
[image]

akusav, 2022-08-04:
[image]

akusav, 2022-08-04:
[image]

tswfi, 2022-08-05:
Nestekaasupulloille on saatu tänään lupa elää rullakossa lastauslaiturilla huoltomiehiltä. Merkintää parannetaan kuitenkin vielä (laput on jo olemassa, yksi ulko-oveen, yksi + nuoli osoittamaan pulloja kohti ja pullojen luoksi yksi).

akusav, 2022-08-05:
[image]

akusav, 2022-08-05:
[image]

akusav, 2022-08-05:
[image]

akusav, 2022-08-05:
https://wiki.tampere.hacklab.fi/pelastussuunnitelma

akusav, 2023-02-28:
Räjähdyssuojausasiakirjaa ei kaiketi tarvita?