TampereHacklab/Inventaario #39: Ripustettavia laitekylttejä

akusav, 2022-08-24:
Hacklabillä kaivataan seuraavia helposti ripustettavia kylttejä (tee ite tai dyykkaa):
- [ ] epäkunnossa
- [ ] huollon tarpeessa
- [ ] puhdistuksen tarpeessa
- [ ] käyttökielto toistaiseksi

Tarkoituksena ilmaista helposti, että laite/kone vaatii huomiota ennen seuraavaa käyttöä mikäli itsellä ei ole juuri nyt aikaa epäkohdan korjaamiseksi.

esim, [image]

314Sami, 2023-10-22:
Jokaisessa kyltissä olisi hyvä olla tussattavaksi kohdat:
______ Jättäjä
______ Päivämäärä
______ Syy

Lisäksi Läbillä on käytössä laminaattori, jolla tulostettavat A4-laput voi laminoida, jolloin ne kestävät vähän pidempään käyttökunnossa.

pinq-, 2023-10-22:
Näihin voisi lisätä vaikka alatekstin "katso githubin issuesta lisätietoa"

GreySkySpin, 2024-04-19:
**English**
I put some cardboard mockups of the Korjaa Minut tags around Hacklab, if they get used I'll put the wooden ones. But I don't want to put the wooden ones until I see how the system is implemented and established. It took time, effort and resources to make the wooden ones, so I'll only put them when/if I feel the community values the Korjaa Minut tags. That means I need to see them in use and that people in the community share photos of things that need to be fixed with the tags.

And can somebody at Hacklab take photos of the Korjaa Minut mockups? I put some on the pegboards in the Lasercutter, Textiles, Electronics and Woodworking areas. Just put the photos in Matrix and I can add them here to the report.

**Suomeksi**
Hacklabin ympärille laitoin Korjaa Minut -lappujen pahvimaketteja, jos ne tottuu niin laitan puiset. Mutta en halua laittaa puisia, ennen kuin näen kuinka järjestelmä on toteutettu ja vakiintunut. Puisten tekemiseen meni aikaa, vaivaa ja resursseja, joten laitan ne vain silloin, kun/jos koen, että yhteisö arvostaa Korjaa Minut -lappuja. Tämä tarkoittaa, että minun on nähtävä ne käytössä ja että yhteisön ihmiset jakavat tunnisteiden avulla kuvia asioista, jotka on korjattava.

Ja voiko joku Hacklabista ottaa valokuvia Korjaa Minut -malleista? Laitoin joitain kiinnityslaudoille Lasercutter-, Tekstiili-, Elektroniikka- ja Puuntyöstöalueilla. Laita vain kuvat Matrixiin ja voin lisätä ne tänne raporttiin.

GreySkySpin, 2024-04-19:
Also here is my list of signs needed. I collected these from chatting with people in Matrix:

CANVA DESIGN FILE
A4 Landscape
https://www.canva.com/design/DAF4i6Jv96E/qa6P0rtfqI7crCiid6Cqzw/edit (request access from me)

Media files: https://github.com/TampereHacklab/Media

Signs we need
- Ota, Ole Hyvä / Free to Take
- Rikki / Broken: https://github.com/TampereHacklab/Inventaario/issues/39
- Rakenteilla olevat kyltit / Under construction

🚧 RAKENTEILLA 🚧
Ei valmis käytettäväksi

🚧 UNDER CONSTRUCTION 🚧
Not Ready for Use

PUUHUONE / WOOD ROOM
- roskakori menee tänne/ waste bin goes here
- polttopuut menee tänne / firewood goes here
- pölynimuri täällä / vacuum cleaner here
- täällä ei koskaan käy mikään / nothing goes here ever

LASERLEIKKURI / LASERCUTTER 
- Muistuta ihmisiä imuroimaan lattia ja tyhjentämään uudelleenkäyttölaatikko työskennellessään siellä / remind people to vacuum the floor and empty the reuse box while working there

MAALAUSHUONEET / PAINTING ROOMS
- Graffiti seinäkyltit ruisku- ja maalaustiloihin / Graffiti wall signs for spray and painting rooms

METAL ROOM 

A "before you leave" poster might also help:
- put away gear
- clean work area
- turn off machines
- shut off gas canisters