TampereHacklab/Inventaario #43: Puupuolen sähköt

pinq-, 2022-11-23:
Puupuolelle tulisi uusia sähköt eli käytännössä lisätä pistorasijoiden paikkaa.
Vaatii suunnitelman ja sähköasentajan. 

- [ ] Alustava karkea suunnitelma
- [ ] Budjetti
- [ ] Tekijä
- [ ] Toteutus

pinq-, 2022-11-25:
[image]
Alustava suunnitelma.

MikkoMyntti, 2022-11-28:
[image]
Kaapelihyllyjen ja valaisimien alustava suunnitelma

MikkoMyntti, 2023-02-16:
[image]

Valaisimista pientä vertailua. Pyydän näistä vielä tarjoukset, kun muut tarvikkeet selviävät. 
https://www.finnparttia.fi/epages/finnparttia.sf/fi_FI/?ObjectID=59035665
https://www.onninen.fi/onnline-teollisuusval-sulj-onnproof-ip66-7000lm-51w-840-1454mm/p/CNI703
https://www.onninen.fi/airam-teollisuusvalaisin-suljettu-ip65-56w-840-pco-gr/p/CFK079?term=4310605
https://verkkokauppa.sonepar.fi/fi/suljettu-teollisuusvalaisin-dp-value-1500-50w-4000k-ip65-4315562

sbeach92, 2023-07-06:
https://www.motonet.fi/fi/tuote/9000992/Brennenstuhl-tyoaseman-roikkuva-virtapiste-8-x-230V-IP20

Näistä taisi olla jossain kohtaa puhetta

Sandplum, 2023-09-07:
Get lamp

tswfi, 2023-09-20:
[image]

Lights have arrived, next up. Get them to the rails and cable them.

Sandplum, 2023-09-24:
Lamps are now mounted on the railing according to the plans shown above. for the "uusi" lights in red.  Plus one was added to the wall in the wood cnc room.  Replacement of old fixtures with new ones will wait until after the new ones are wired and can provide work light for removing the old fixtures.

The lamps now await electrical wiring, preferably performed by (or supervised by)  a licensed electrician.

We've made a lot of progress and are so close now.  Let's see if we can push this through and enjoy having proper lighting in the Wood Workshop.  :sunglasses:

Sandplum, 2023-10-21:
Mikko will supervise the pulling of cables and connecting the lamps on **Saturday, October 28th**.  At the same time, we can run a new supply cable.  It is heavy and requires help from **a lot of people**.

tswfi, 2023-10-28:
Uusia valoja kytketty, syöttökaapelia vedetty. Homma etenee!

Sandplum, 2023-11-10:
Old lamps were removed and new lamps are waiting to be hung in those places.

nuulis, 2025-02-09:
Uudet valaisimet asennettiin talvella 2023-2024. Pistorasiat (pl. voimavirta) on asennettu seiniin syksyllä 2024. Puuttuu ns. enää loput