TampereHacklab/Inventaario #44: Hihnatulostin ei pelaa. Creality CR-30 3DPrintMill

pinq-, 2022-11-25:
Tulostus ei jää petiin kiinni. Voi vaatia paljon säätöä.
Malli: Creality CR-30 3DPrintMill

pinq-, 2023-07-11:
Tulostin tarvitsee uuden virtalähteen. 
Todennäköisesti tämä malli: https://www.3dprima.com/spare-parts-accessories/manufacturers/creality-3d/creality-3d-cr-30-power-supply_27009_8454

tswfi, 2023-07-12:
suomalainen source https://www.an-cadsolutions.fi/tuote/creality-3d-cr-30-virtalahde/ (kiitos Arno löytämisestä)

Jos olisi tiedossa koko ja voltit ja ampeerit niin voisi myös katsoa tipahtaako tuollainen jostain syliin.

Tosin mietin myös että kannattaako tuohon tuota 40€ investoida jos on muutenkin tarttumisongelmaa pedissä? Jos laittaisi vaan myyntiin ilman poweria?

pinq-, 2023-09-18:
Virtalähde laitettu tilaukseen

pinq-, 2023-10-11:
Virtalähde asennettuna ja toimii. Vaatii säädön kohdilleen

closed 2023-10-11 by pinq-