TampereHacklab/Inventaario #49: sopivien palovaroittimien hankinta ja asennus

akusav, 2023-02-28:


Ripae, 2023-07-05:
- [ ] Palovaroittimien hankinta, 10-15 kpl
- [ ] Varoittimien muutostyö kaapelilla ketjutettaviksi
- [ ] Sijoituspaikkojen valinta, kaapeloinnin suunnittelu 
- [ ] Kattokiinnikkeiden asennus
- [ ] Kaapelin ja liittimien valinta/hankinta
- [ ] Kaapelikiinnikkeiden, asennusputkien ym. hankinta
- [ ] Kaapelin asennus
- [ ] Kuittauspainikkeiden hankinta ja asennus
- [ ] Sähkönsyötön suunnittelu
- [ ] Virtalähteen/laturin ja akku-/paristovarmistuksen osien hankinta (, rakentaminen, kotelointi) ja asennus
- [ ] Home Assistant -liitännän suunnittelu
- [ ] HA -liitäntäosien hankinta ja asennus
- [ ] HA:in konffaus ja muut softahommat

nuulis, 2025-02-09:
Varoittimia ja kaapelia on asennettu, mutta onko systeemi kuinka lähellä toimintaa?

akusav, 2025-02-09:
Tekijästä ei taida olla mitään tietoa?

314Sami, 2025-03-08:
Projektin alkamisen aikaan aktiivisin tekijä taisi olla uudehko jäsen ja oletus/tahtotila oli saada tästä "yhteisön projekti" ja kovin moni ei jostain syystä osallistunut. En silti näe mitään syytä olla jatkamatta projektia, tärkeä asia ja aika pitkällä kokonaisuudessaan.