TampereHacklab/Inventaario #63: Jauhemaalaus käyttöönotto

tswfi, 2023-09-10:
- [ ] Maalauspaikan valmistelu
- [ ] Säilytyspaikka sekä itse maalaimelle että maalijauheille
- [ ] Käyttökoulutus / ohjeet (ceravyn aloittanut jo)
- [ ] Paineilmaliitin. Mukana tuli se british pieni standardi, tarttis saada niille kierteille sopiva pikaliitinadapteri
- [ ] Maadotuspiuha. Mukana tuli sellane onneton "tällä pääset alkuun" mut paljon parempaa jälkeä saa kun maadottaa kunnolla johonki pistorasian maahan tms. Ehdotus että vedetään kevi piuha rasiaan lähelle maalauspistettä, siihen banaani liitin ja sopiva välijohto hauenleuka johdolla kappaleeseen (esim silikoni johtoa niin on kiva käsitellä)

Ceravyn, 2023-09-21:
Ruuvasin banaaniliittimellä varustetun rasian seinään ja vedin keltavihreän seinää pitkin ylhäällä olleeseen pistorasiaan hauenleualla.

Olis kiva jos sais jonkun kiinteemmän/nätimmän asennuksen kevi-johdolle, mutta nykyinen oli parasta mitä illasta keksin ja koska se toimii, niin todennäköisestihän se jää tuollaiseksi seuraavaksi X vuodeksi :)

[image]

Ceravyn, 2023-10-07:
Lisäsin issueen pysyvän voimavirtajaon tarpeen tulityöpuolelle uunia varten. Uunia on toistaiseksi käytetty irrottamalla migi ja kytkemällä uuni tähän pistokkeeseen, mutta tämä ei toimi jos toiset haluavat hitsata samaan aikaan. Viimeksi vedin jatkoroikan tulityötilan ulkopuolelle ja irrotin pylväsporakoneen seinästä ja tökkäsin uunin sen pistokkeeseen käytön ajaksi.

Ceravyn, 2023-10-07:
Vahinkoklikkauksena "closed issue" :)

tswfi, 2023-10-07:
Voimavirta haaroitettu

nuulis, 2025-02-09:
Jauhemaalaus on otettu käyttöön ja toimii suunnitellusti. Mahdollisista parannusehdotuksista voi tehdä uuden issuen.

closed 2025-02-09 by nuulis