TampereHacklab/Inventaario #65: Large-format HP Designjet 4000 Printer doesn't print.

Sandplum, 2023-09-24:
Something something print heads...

tswfi, 2023-11-10:
From Otto: Look for HP designjet 4000 printhead kits.  We were talking with tsw about replacing all. There's two set of printheads in the machine

tswfi, 2023-11-10:
https://support.hp.com/au-en/document/c00263211 says these are the product numbers

[image]

tswfi, 2023-11-10:
and cleanup procedure from hp https://support.hp.com/nz-en/document/c00264815

Sandplum, 2023-11-10:
[image]
[image]
[image]

tswfi, 2024-02-07:
Spent 40 minutes on the phone with HP support about this.

This printer is so old that there are no replacement parts for it anymore.

New printheads *might* help but they have no idea (as it does not seem to read the serial number it might be something else entirely) 

https://www.dustin.fi/product/5010077707/printheadcleaner-no90-syaani-till-dj-4000 296,98€ for _one_ printhead and it requires both to work... (and probably no return policy for these)

sbeach92, 2024-02-13:
https://h30434.www3.hp.com/t5/Printers-Archive-Read-Only/printhead-appears-to-be-missing-not-detected-or-incorrectly/m-p/8634740/highlight/true#M3377566

Found this... Would try to check later today if there is some web features enabled if some sort of reset would wake it up.

sbeach92, 2024-02-14:
https://www.ebay.com/itm/225163150719

brndd, 2024-02-27:
Cyan printheads swapped. During calibration printer showed no error but leaked black ink everywhere.

brndd, 2024-02-29:
Black ink problem probably caused by the fact that the black printhead cleaners had had their cleaning sponge removed and each were overflowing with black ink. All the overflow was getting sprayed on the other printhead cleaners, on the printheads, and inside of the machine.

Every printhead and printhead cleaner needs to be replaced, excepting maybe the cyan ones which were just replaced, unless they already became soaked with black ink. The printhead cleaners are keyed to the color (not sure there is any real reason to do this other than to inconvenience users) with a physical key on the cartridge.

After that the entire machine needs to be cleaned because there is black ink **everywhere**.

Replacement print heads will cost ca. 200€ if buying out of date stock from eBay. Cleaning the machine will probably take a full day and be frustrating and messy. Lugging the machine to the loading dock and gently pushing it off will take 15 minutes.

nuulis, 2024-03-06:
Yhteisömiitissä ti 5.3.2024 todettiin laite toivottomaksi korjata ja päätettiin laittaa printteri Toriin/vastaavaan "annetaan"-osastolle. Joku saattaa tarvita varaosiksi. Jos ei noin kuukauden sisään tärppää, kone viedään serriin.

Community meeting 5.3.204 decided that machine is too broken to be fixed. It'll be put on Tori.fi "for free" and if nobody wants it in about a month we'll discard it as e-waste.

tswfi, 2024-07-22:
This printer has been decommissioned. Closing the issue

closed 2024-07-22 by tswfi