TampereHacklab/Inventaario #77: Ultraäänipesurissa vikaa

tswfi, 2023-10-27:
Otto raportoi kanavalla että UÄ pesuri ei pelitä.
Pitää ääntä mutta tehot on kateissa.
Olisiko driveri kuollut?

Ceravyn, 2023-12-09:
UÄ pesuri siirretty maalaamon räjäytyksen tieltä pois SER-jätteen eteen. Lappu päällä "Fix me or throw away".

tswfi, 2023-12-14:
[image]

[image]

[image]

[image]

[image]

Miggov, 2023-12-18:
tsw:
@Miggov: https://github.com/TampereHacklab/Inventaario/issues/77 tuolla kuvia sisältä. taisi olla 8 transduceria ja neljä driver boardia (jokainen ajaa kahta). mitä katselin hintoja niin 2x transducer + driver oli ~60€ kiinasta

uusi kone maksaa muutaman satasen

Mikko V:
Joo kattelin gh mut vähän tosiaan jäi kysymyksiä

Oishan toi sitten vähä halvempaa korjata mutta millä lie maailmanlopun liimalla noi on kiinni

Yks boardi oli vaa kärähtäny ja noi kuvien transducerit?

tsw:
en mitannut boardeja, en tiedä niiden kunnosta

lähinnä tuo diodin vieressä oleva roska oli kuvan kohteena

Mikko V:
Niijjoo, silmämääräisesti meinaan

Joo katoin et jotain ruosteveden näköstä siin kanssa

tsw:
joo siellä oli aikapaljon roskaa sisällä

mun tuomio on että seuraava joka tarvitsee käy ensin kaupassa :D

nuulis, 2025-02-09:
UÄ-pesuri on viety serriin. Ei voitu korjata

closed 2025-02-09 by nuulis