TampereHacklab/Inventaario #84: Power 2000 smirgeli metalliharja irtoamassa

tswfi, 2023-11-24:
Harjan kiinnitys korjattava. Kone käyttökiellossa.

tswfi, 2023-11-24:
[image]

Sproketti, 2023-12-05:
Kurkkasin tuota konetta, mutta en keksinyt miten korjata. Tuossa on joku erikoinen viritys teräsharjan puolella ja akselit ovat erilaiset. Olisko joskus tuo toisen puolen akseli hajonnut vai pitäisikö tuohon olla joku sopiva kiinnitys mekanismi?

[image]
[image]

nuulis, 2025-02-09:
Tilanne on edelleen sama. Vasempaan akseliin pitäisi ehkä sorvata tiukka sovitepala teräsharjaa/kiillotuslaikkaa varten.