TampereHacklab/Inventaario #96: Repair blue canister vacuum cleaner

Sandplum, 2023-12-13:
Blue canister vacuum cleaner is not functioning (reported by Arno and Teemu).

Once repaired, return the vacuum cleaner to the laser cutter corridor.

tswfi, 2023-12-15:
Nilfisk Alto

Cable was broken near the socket. Fixed.
Power switch was acting up. Feels fixed now.
Motor overheats. Not possible to open. Replacement part https://www.huoltopalvelu.com/Poelynimurin-moottori-Domel-4673402-5

tswfi, 2023-12-15:
And it seems that the suction power control is at full blast all the time, no one wants to run this at half speed so no need to debug it deeper...

tswfi, 2023-12-20:
tsw orders new motor in a moment

tswfi, 2023-12-20:
Motor ordered, wont probably get here before christmas though...

tswfi, 2023-12-23:
Working again

[image]

closed 2023-12-23 by tswfi