TampereHacklab/Inventaario #99: Ei toimi Fu03

Ottopoeka, 2024-01-25:
Taajusmuuttaja antaa virhettä Fu03 eikä anna liikuttaa akseleita.  Kenties huonoa sähköä? Karli ja Arno ovat jokseenkin tietoisia ongelmasta.

spiikki, 2024-01-25:
Toimii satunnaisesti. Taajuusmuuntaja virtakipuilee.

tswfi, 2024-01-25:
Mikä tamu siinä on?
Olisiko kuvaa?

Ottopoeka, 2024-01-25:
[image]

sbeach92, 2024-02-06:
[image]

Ylivirtaa, voi johtua esim alijännitteestä. Enable päällä ennenku tamussa virrat? Huono jänniteen syöttö.

https://wiki.tampere.hacklab.fi/_media/tyokaluja/labcncvfd.pdf

pinq-, 2024-02-07:
Oisko mahkuja kokeilla toista tamua? Minkälaisen se vaatisi?

Ottopoeka, 2024-02-15:
[image]

Ottopoeka, 2024-02-15:
Vaihejännite oli mitattuna about 390. Muutin parametrin "FC.0 Under voltage protection level 360～460V" arvosta 380 370:een ja virhe katosi

closed 2024-02-24 by Ottopoeka