TampereHacklab/Inventaario #100: Lathe tailstock locking mechanism is broken

Sproketti, 2024-01-28:
The locking mechanism for lathe tailstock is broken and there is a bolt currently in place to keep the machine usable.

The bolt in the mechanism is hold in place by pin that breaks easily if too much force is applied. I fixed this a couple of months ago, but did not have time to take care of it right now.

Is there a need to make stronger mechanism or is this weak pin in place by design to protect other parts of the lathe?

[image]

[image]

[image]

tswfi, 2024-01-28:
I'm guessing it is to protect other parts (or the threads).

Not sure if these would work https://www.puuilo.fi/hercules-putkisokka-nippelisarja (I will bring a set anyway)

Sproketti, 2024-01-28:
I was thinking the same. I have those pins in my smartstore box if someone wants to tackle the issue before I have time.

tswfi, 2024-01-28:
Hain sokat otto asensi, toimii taas

[image]

closed 2024-01-28 by tswfi