TampereHacklab/Inventaario #102: Jauhemaalausuuni

Miggov, 2024-02-06:
Teline maalattaville pitäs hitsata ja asemoida uuni maalaustilaan

314Sami, 2024-02-06:
Aiemmin säädetty mm.:
- 3-vaihetuulettimet kytketty rinnan taajuusmuuntajan taakse 6 Hz vakionopeudelle (potikan lisääminen mahdollista yms)
- palanut lamppu uunista ehjätty, varalamppu sähkötilan sisällä
- lämmitysvastukset kytketty siten, että ylätuulettimella on 2 kpl ja alatuulettimella 1 kpl käytössä, tehot yhteensä n. 12 kW
- vedenlämmitin/höyrykone kytketty pois toiminnasta, vesipumppu myös

314Sami, 2024-02-06:
Puhallinten nopeus säädetty 8 Hz ja vastusten kytkentä vaihdettu siten, että alhaalla kaksi ja ylhäällä yksi vaihe. Näin lämpötila jakaantuu uunissa tasaisemmin.

Tehtävää: jos moottorien sirinä ei miellytä korvaa, tamun asetuksista voi koittaa säätää kytkentätaajuutta.

Miggov, 2024-02-21:
Tai ehkä paras käyttää hitsipöydän alta löytyvää isoa suorakulmaa jolla kattoo pystyjen suoruuden suhteessa renkaisiin

Miggov, 2024-03-03:
Kärri vailla vielä pellin sovitusta ja hitsausta estämään ilman falskaaminen uunin alaosasta ja sopivien telineiden tekoa. Suunnitletiin että semmoinen vähän uuniritilää mukaileva "kalteri" tms voisi olla aika yleiskäyttöinen pikkuosien kanssa. Hyllyn sakaroihun näitä ei kuitenkaan tule hitsata ettei mene monikäyttöisyys

Miggov, 2024-03-03:
Ja sit jokku ristituet noille pystytangoille ois nice painavampien maalattavien kanssa

tswfi, 2024-05-07:
Sähköt kunnossa

nuulis, 2025-02-09:
Jauhemaalausuuni on otettu käyttöön ja toimii suunnitellusti.

closed 2025-02-09 by nuulis