TampereHacklab/Inventaario #103: Tasopiirturi Roland DPX-3300

tswfi, 2024-02-07:
Nykyinen softa ei ole kovin käyttäjä ystävällinen ja sen ajaminen ei ole kivaa. Voisi katsoa sille paremman paikan ja tarkistaa onko sen ohjelmistoa tai alustaa mahdollista/helppo päivittää. Piirturin säilyttämiselle oli enemmistön puoli

tswfi, 2024-02-07:
about first google results:
https://github.com/xHain-hackspace/PlotterFeeder

could we wall mounted?
https://wiki.bristolhackspace.org/equipment/mainroom/plotter

pinq-, 2024-02-07:
Plotterin huoltoon löytyy myös blogi postaus:
http://anachrocomputer.blogspot.com/2013/04/fixing-big-plotter.html

tswfi, 2024-02-07:
ja täydet schemat https://archive.org/details/dpx-3300-schematics

brndd, 2024-02-16:
Napeissa huonot kontaktit. Tartteis irrottaa piirilevy ja näyttää puhdistusainetta.

3D-tulostettu kynänpidike vähän liian jäykkä jonka takia työkalunvaihto toimii vaihtelevalla menestyksellä. Työkalupidikkeissä on samoja leukoja, mitä kelkassa, joten vaihdetaan kelkan leuat yhden työkalupidikkeen kanssa päikseen ja katotaan toimiiko paremmin. Toinen vaihtoehto olis hioa 3D-mallia kunnes se on just eikä melkeen alkuperäistä vastaava, nykyinen (joka löytyy Nextcloudista) on yhden iltapäivän aikana väsätty.

nuulis, 2025-02-09:
Piirturi on "printtausnurkan" uudelleenjärjestelyn yhteydessä irroitettu jalustastaan, ja suunniteltu nostettavaksi seinälle "museoon"