TampereHacklab/Inventaario #110: Sorvin moottori käryää. Sorvi poissa käytöstä.

Sproketti, 2024-03-10:
Sorvi alkoi kesken kaiken haista pahalle ja savuta. Pikaisen tutkinnan jälkeen vika paikannettu moottoriin, josta savu ja käry lähtee.

[image]

[image]

[image]

[image]

Sproketti, 2024-03-10:
Moottori purettu irti ja todettu vaativan uudet hiilet. Ei varmuutta onko tilattu, mutta aiheesta keskusteltu asiakanavalla 10.3.2024.

Puretut osat ja irroitetut johdot merkattu. Koneen päälle laitettu lappu, joka kertoo koneen olevan käyttökiellossa.

[image]

[image]

[image]

tswfi, 2024-05-01:
Uusi moottori laitettu tilaukseen.

tswfi, 2024-05-10:
Uusi moottori saapunut

tswfi, 2024-05-14:
Uusi moottori asennettu ja kytkennät kunnossa.

Vain hitaampi alue käytössä (kiilahihnapyörä ei mahdu tarpeeksi syvälle). Tehdään tuosta uusi issue koska ei estä käyttöä.

closed 2024-05-14 by tswfi