TampereHacklab/Inventaario #120: Peräkärryn vetokita väljä ja töpseli risa

Miggov, 2024-08-30:
Vetokita on erittäin väljä, voisi vaihtaa ennenkun sattuu enemmän vahinkoja, kerran jo auennut.

Töpseli toimii mutta siitä puuttuu pala. Voisi vaihtaa myös.

Miggov, 2025-07-23:
> Peräkärryn kuulakytkimen väljyydestä ei ole tullut enempää palautetta.
> Valojen liittimestä puuttuu edelleen pala, mutta ei vaikuta valojen
> toimintaan.

Tuo on yhden auton koukusta irronnu ajossa et kyl se ois varmaan jo
asiallista vaihtaa eikä oo monen markan juttu jos vaa saa suorittaa. Kuluva
osa ja tossa ei ole kulumisindikaattoria, tosin kahva menee ihan pohjaan
joka sanois kovasti et alkaa olla lopussa

tswfi, 2025-09-01:
Vedin tuota kärryä juuri viinikkalafestejä varten ja kita tuntui ihan ok:lta ja valot pelittivät ok.