TampereHacklab/Inventaario #128: Combine each Finnish wiki page and its English page into one page containing both languages.

Sandplum, 2025-01-28:
Often the Finnish wiki pages are updated without also updating the English page.  Editors of the Finnish pages often do not know there are English pages.  Readers of the English pages do not know that the pages do not contain the most recent information.

A single page, containing the information in both languages, allows editors and readers to see all of the information and make comparisons, and if necessary, use their own translator.

sbeach92, 2025-03-07:
I would like to see this plugin working: https://www.dokuwiki.org/plugin:translation so works like https://www.dokuwiki.org/dokuwiki

314Sami, 2025-03-08:
> I would like to see this plugin working: https://www.dokuwiki.org/plugin:translation so works like https://www.dokuwiki.org/dokuwiki
Then everyone that know at least 2 languages can't see the info in both languages to see if the info in both of them matches. In other words, the translating will be done only when people are actively looking for something to translate, not like "oh, that is not up-to-date, lemme just tune it real quick..."

sbeach92, 2025-03-08:
If you go to https://www.dokuwiki.org/dokuwiki and change to some other language you probably see banner that this translation is holder than current main page and get allso diff link between versions of Current page and one that translation is made against.

I see it much more convient to keep on track if translation is up to date than reading trough both language versions.

I also think that dual language pages can easily led to situatin where limited english skill are felt as an barrier to update wiki.