TampereHacklab/Inventaario #131: Plasma cutter malfunction.

Sandplum, 2025-03-08:


Ceravyn, 2025-03-12:
[image]

Wiring for reference.

[MESA7i76HL.pdf](https://github.com/user-attachments/files/19209392/MESA7i76HL.pdf)

Ceravyn, 2025-03-19:
Plasma-CNC's new stepper control card has been crowdfunded, ordered, delivered and connected to the CNC table. The old desktop computer which used to run plasma has been retired and a new Lenovo All-in-one PC installed to its place.

- New PC also comes with integrated wifi so sending gcode files via network is now possible as well.
- New PC looks better and has more accessible USB ports and overall just a nice upgrade.
- Newest version of LinuxCNC -image has been installed on the new PC. UI and some configuration are slightly different from the previous one, maybe has some more functionality too. If you've old gcode files for batch cutting, they might not work the same way with the new config, so check before cut!

The new control card has an integrated FPGA and doesn't require the separate FPGA-card inside the old desktop PC. However that card is still valuable and should be taken into safekeeping before completely discarding the old PC.

- [x] Backup configuration from the old PC
- [ ] Take the PCI-card into safekeeping
- [ ] Bonk the old PC and monitor

Plasma CNC is still out of service until:

- [ ] LinuxCNC configuration is done and working. Currently the link is established, card is recognized, but LinuxCNC throws error message when trying to move the toolhead.
- [ ] Plasma/compressor room and the immediate surroundings of the control PC (AIO) have been cleaned and things put into their places. If things don't have a place, places needs to be invented.
- [ ] New holder for the new stepper card to be lasercut and installed into place. Currently the card is **not** attached to anything and is just laying on top of plastic sheet for testing purposes.
- [ ] Old 25-pin serial cable removed, new ethernet cable wired to its place securely without tripping hazards.
- [ ] CNC table wiring EMI-shielding and grounding to be improved.

Same crowdfunding also acquired a THCAD2 high isolation module, which allows for height control of the plasma torch (Z-axis). **This is a separate module and not required for operating the CNC** but will allow for more and new tricks for the machine. Installing it can be done while upgrading the stepper card, wiring, etc. or be left for the future. This is still undecided.

- [ ] THCAD2 installed to place
- [ ] THCAD2 configured to LinuxCNC

¯\_(ツ)_/¯